hku-mars/r3live #189: Get pointcloud data from ros messages fail

hankge, 2023-07-31:
Thanks for your work, I am very excited to try it!
I am using Ubuntu 20.04 + ROS noetic + Ouster os1-64 lidar with inner imu data + zed2i camera.
I was trying to start the r3live_bag_ouster.launch file to do mapping, but when I started to run bag, I received the error message related to the point cloud. See attached below. 
[image]

Could someone tell me how to fix this error? 
Thanks in advance.

alwynmathew, 2023-08-02:
For some reason, it only works for lidar scans with `fields.size() < 3` or `fields.size() == 8  && fields[ 3 ].name == "intensity" &&
fields[ 4 ].name == "normal_x"` or `fields.size() == 4 && fields[ 3 ].name == "rgb"` as shown below: https://github.com/hku-mars/r3live/blob/6143a38537f28cb36eb24e9bbe2e39c8f7967157/r3live/src/r3live_lio.cpp#L91

hankge, 2023-08-02:
Hi @alwynmathew , I also seen this code. Here is my lidar topic output.header: 
  seq: 4430
  stamp: 
    secs: 528
    nsecs: 948532131
  frame_id: "os_sensor"
height: 64
width: 2048
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "intensity"
    offset: 16
    datatype: 7
    count: 1
  - 
    name: "t"
    offset: 20
    datatype: 6
    count: 1
  - 
    name: "reflectivity"
    offset: 24
    datatype: 4
    count: 1
  - 
    name: "ring"
    offset: 26
    datatype: 4
    count: 1
  - 
    name: "ambient"
    offset: 28
    datatype: 4
    count: 1
  - 
    name: "range"
    offset: 32
    datatype: 6
    count: 1
is_bigendian: False
point_step: 48
row_step: 98304

I have no idea about knowing my lidar(ouster os1-64) field size is suitable or not. How can I get the field size of my lidar?

alwynmathew, 2023-08-03:
Your LiDAR field size is 9, as shown in the error screenshot above. A workaround is adding the below statement after L112
```
if ( ( msg->fields.size() == 9 ) ) 
        {
            pcl::fromROSMsg( *msg, pcl_pc );
            return true;
        }
```
This should make the package work, but still, you will get a warning that normals are not found. They expect the input to be in the `PointXYZINormal`. Good luck!

hankge, 2023-08-03:
Thanks for your help! As you imagine, the point cloud is shown in Rviz, but the "normal_x" "normal_y" "normal_z" are missing.
[image]
I'm trying to change PointXYZINormal to PointXYZI, but I don't know if that works.

closed 2023-08-23 by stale[bot]

zxhasd, 2023-11-29:
> 感谢您的帮助！正如您想象的那样，点云显示在 Rviz 中，但缺少“normal_x”、“normal_y”、“normal_z”。 我正在尝试将 PointXYZINormal 更改为 PointXYZI，但似乎不起作用。 ![图像](https://private-user-images.githubusercontent.com/88472118/258221184-b6bfa320-9c87-49bc-b689-559422e36dba.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEyNjA2MDIsIm5iZiI6MTcwMTI2MDMwMiwicGF0aCI6Ii84ODQ3MjExOC8yNTgyMjExODQtYjZiZmEzMjAtOWM4Ny00OWJjLWI2ODktNTU5NDIyZTM2ZGJhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI5VDEyMTgyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJlYjgxNmJmYTIzZDkxZDRlZGJlODMwOTI3MmVkNDc2NjI4NTg0NGVjYTMzZjRlZTRjYmU1OTRhN2FiMzRhOGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.imGsAlET8Dv_s-oDCinzezfq19DHKdHU7WZ5pc895bM)

请问您解决这个问题了吗