hku-mars/r3live #195: Is it possible to use Livox Mid 360 (a spinning lidar) and two cameras on two side of the lidar and use the algorithm to colorize 360 degrees data?

oscarpoudel, 2023-10-11:
I tried modifying the code to take into account the whole 360 degrees of the lidar but could not find a way to do so. And I cannot buy a new solid state lidar too so wanted to employ two cameras on two ends of the spinning lidar and try to make 360 RGB point cloud at once.

li100102, 2023-10-17:
Hi Oscar,

It is possible to run r3live with Livox mid 360, however, you need to make some modification with the source code.
In "loam/LiDAR_front_end.cpp" line 219, function horizon_handler has judged the points with x < 0.7 as an invalid data. This does not fit for a spinning lidar.
https://github.com/hku-mars/r3live/blob/6143a38537f28cb36eb24e9bbe2e39c8f7967157/r3live/src/loam/LiDAR_front_end.cpp#L215C1-L220C10
Thus, you should remove the condition "msg->points[ i ].x > 0.7" and use lidar type 1 in your config. This should work for your platform.

oscarpoudel, 2023-10-17:
Thank you for the help. Now I can gather all 360 degrees of point cloud with my lidar. 
[image]

closed 2023-10-17 by oscarpoudel

YWL0720, 2023-10-23:
Hi, may I ask how you managed to synchronize the point cloud messages published by mid360 with the image messages? Since mid360 uses livox_ros_driver2 not livox_ros_driver@oscarpoudel

BellosC, 2024-02-09:
@oscarpoudel 

Hello!
Can you please inform us if you managed to get RGB pointcloud using mid360 and camera or cameras?

peter-ivarsen, 2024-06-26:
Had the same problem with the Mid360. 
For me it worked to set lidar type to 0 in the config file, (i.e. to lidar type is MID40)
Then I had to change the default paramter in [line 98](https://github.com/hku-mars/r3live/blob/6143a38537f28cb36eb24e9bbe2e39c8f7967157/r3live/src/loam/LiDAR_front_end.cpp#L98)  in LiDAR_front_end.cpp from 0.1 to -500, (i.e. all points with x value less then -500 are keept.) This is due to a filtering operation done in give_feature()