hku-mars/r3live #204: Error accessing the Google Drive/MS OneDrive Link for the Dataset

sumedhfb, 2024-03-27:
The dataset links are not accessible. It returns a 404 error. Can you please let know if there is any other source (accessible from the US)

ziv-lin, 2024-03-29:
Sorry, I have noticed this issue too. Since I have graduated from school, the shared folder is no longer available. I apologize for any inconvenience this technical issue might cause, and I will fix this problem as soon as possible.

sumedhfb, 2024-03-29:
> Sorry, I have noticed this issue too. Since I have graduated from school, the shared folder is no longer available. I apologize for any inconvenience this technical issue might cause, and I will fix this problem as soon as possible.

It would really help if I could access the data, thanks a lot!

ziv-lin, 2024-04-03:
Hi, I have just moved the old bag to a new account now, could you please check if is following link can be accessed normally?
https://1drv.ms/f/c/3e715b7aa136191a/EqpK7QnN4OpCqHmL2ykpZ50Bjz3pyJ0kwyvwpBBLtzR4bQ?e=TqPN8E

sumedhfb, 2024-04-04:
> Hi, I have just moved the old bag to a new account now, could you please check if is following link can be accessed normally? https://1drv.ms/f/c/3e715b7aa136191a/EqpK7QnN4OpCqHmL2ykpZ50Bjz3pyJ0kwyvwpBBLtzR4bQ?e=TqPN8E

Thank you for this! I was able to test R3Live and the bag successfully. However, there does not seem to be documentation about using Velodyne with R3Live. All the other repos - Fast-Lio, Fast LIVO etc have a velodyne implementation option in them. Is there a reason why it does not exist in R3Live. If possible, can you let me know how to get it working with Velodyne? I am using a Velodyne Puck Hi-Res (VLP16) and a Zed2 Camera

closed 2024-04-21 by stale[bot]