hku-mars/r3live #220: Bad result in MARS dateset

XuMeng9612, 2024-08-26:
I want to test r3live in MARS dataset (https://mars.hku.hk/dataset.html), since r3live show great result in paper "MARS-LVIG dataset: A multi-sensor aerial robots SLAM dataset for LiDAR-visual-inertial-GNSS fusion".
However, the result is poor in dataset HKairport01 as below, which is even worse in HKairport03:
[image]
This is my config file:
Lidar_front_end:
   lidar_type: 1   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 6
   using_raw_point: 1
   point_step: 1
   
r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 4  # The point step of append point to global map. [default = 4]

r3live_vio:
   image_width: 2448
   image_height: 2048
   camera_intrinsic:
      [1451.28, 0.0,       1177.5,
               0.0,     1451.29,  1043.5,
               0.0,     0.0,      1.0] 
   camera_dist_coeffs: [-0.0572, 0.1209, 0.00124, -0.00018, -0.06327]  #k1, k2, p1, p2, k3

   camera_ext_R:
      [0.0032,-0.0056,0.9997,
      -0.9997,0.0227,    0.0045,
      -0.0229,   -0.9994,   -0.0559]
   camera_ext_t: [0.0317,0.0008,0.0748] 
   
r3live_lio:        
   lio_update_point_step: 4   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: 0        # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30

Is there anything wrong?

closed 2024-09-15 by stale[bot]

JackCooper666, 2025-02-28:
Hi, have you solve the problem ? I got the same issue as well.

liubonan123, 2025-07-24:
Hi, have you solve the problem ? I got the same issue as well.

XuMeng9612, 2025-07-24:
> Hi, have you solve the problem ? I got the same issue as well.
Unfortunately, I didn't solve this problem