hku-mars/r3live #225: ARM device compilation

pereira90-ai, 2024-12-01:
Hi, Jiarong.

I hope you are doing well.

During the installation I got this issue:
C++: error: unrecogniaed command line option: '-msse2'

I think this issue is caused because of the difference between device architecture. (I'm using Jetson Orin whose arch is arm64)
I found those optimization options are not supported on ARM64 device.

And then what should I do? Just eliminating them or should I add new options?

Thanks for your consideration.
Best,

shallwing, 2024-12-23:
> Hi, Jiarong.
>
> I hope you are doing well.
>
> During the installation I got this issue: C++: error: unrecogniaed command line option: '-msse2'
>
> I think this issue is caused because of the difference [...]

I removed the two options "-msse2" and "-msse3" in CMakeLists.txt, then this error you have encounted was disappeared:
`
##  CMakeLists.txt, line 6
set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -std=c++14 -O3 -lboost_system -pthread -Wenum-compare -DWITH_GTK_2_X=ON")
`

pereira90-ai, 2024-12-23:
> CMakeLists

Hi @shallwing 
Thanks for your help.
After removing those options I encountered new issue related to 'cpuid.h'

And currently ROS uses opencv 4.5.4 but I think 4.5.3 is supported.
How can I downgrade this opencv version into 4.5.3?
Thanks.

shallwing, 2025-01-06:
> > CMakeLists
> 
> Hi @shallwing Thanks for your help. After removing those options I encountered new issue related to 'cpuid.h'
> 
> And currently ROS uses opencv 4.5.4 but I think 4.5.3 is supported. How can I downgrade this opencv version into 4.5.3? Thanks.

[https://github.com/hku-mars/r3live/issues/32](problem explaintion)
The author has declaimed in this URL that now we can only run the project in x86...   what a pity!

closed 2025-01-24 by stale[bot]

Slam-journey, 2025-02-16:
Fast-LIVO2 can be directly compiled and run in real time on ARM platforms, such as RK3588, but R3LIVE performs poorly in terms of memory usage and efficiency. You might have a try.

shallwing, 2025-02-19:
> Fast-LIVO2 can be directly compiled and run in real time on ARM platforms, such as RK3588, but R3LIVE performs poorly in terms of memory usage and efficiency. You might have a try.

You are right. After reading that issue, finally I ran FAST-LIO instead of R3live on NVIDIA platform.